Eddddddddy/Songguo-PTS200 #1: firmware instructions

VadimP72, 2023-04-09:
Where can I download the firmware instructions? The link on the firmware page does not work.

Eddddddddy, 2023-04-12:
Not ready yet, sorry.

There is a simple instruction:
1. Long press the confirm button, enter setting menu.
2. Select Firmware Update.
3. Using USB cable connect PTS200 to PC/macOS.
4. Drag new firmware .bin into USB disk.
5. PTS200 will automatic reboot.

If you want to develop your own firmware, development environment is Arduino for ESP32. Confirm button connect to GPIO0 (using for Boot mode).

If you have any question or suggestion, pls leave message here.

VadimP72, 2023-04-12:
Hello! What kind of soldering iron are we talking about? There is no firmware update item in the menu of my soldering iron. Now I have version 3.14. My soldering iron is flashed only through the SP 32 firmware program on the computer. The soldering iron as a USB disk is not detected in the computer.

> 
> 
>Not ready yet, sorry.
>There is a simple instruction:
>*  Long press the confirm button, enter setting menu.
>*  Select Firmware Update.
>*  Using USB cable connect PTS200 to PC/macOS.
>*  Drag new firmware .bin into USB disk.
>*  PTS200 will automatic reboot.
>If you want to develop your own firmware, development environment is Arduino for ESP32. Confirm button connect to GPIO0 (using for Boot mode).
>If you have any question or suggestion, pls leave message here.

VadimP72, 2023-04-12:
[image]
Its my solder.

quotschmacher, 2023-04-18:
inside the arduino ide, which board should i select?

VadimP72, 2023-04-18:
I do not use arduino, but I flash through the regular ESP 32 firmware. What is the firmware procedure ?

VadimP72, 2023-04-19:
I use ESP flash_download_tool_3.9.4 for firmware. Through this program, firmware 4.0, 4.1 and 4.2.1 are flashed, but the soldering iron does not turn on with them and does not work. Maybe they need to be redone somehow?

VadimP72, 2023-04-19:
I use ESP flash_download_tool_3.9.4 for firmware. Through this program, firmware 4.0, 4.1 and 4.2.1 are flashed, but the soldering iron does not turn on with them and does not work. Maybe they need to be redone somehow?

> 
> 
>inside the arduino ide, which board should i select?

Eddddddddy, 2023-04-19:
@VadimP72 Hi, if it's not working, hardware will be the first version (maybe). They are different design. You can try this [link.](https://github.com/Eddddddddy/SolderingPen_FW) Generally they are same.

sbn-purchark, 2024-01-01:
> When I download the 4.5.1 zip it has two files factor.bin and user.bin Any rename or something?

Try user.bin instead. I read somewhere to not use the factory.bin. Also, the 4.5.1 factory.bin is broken, 4.5.0 is fine though.