Eddddddddy/Songguo-PTS200 #4: How to reduce its sensitivity

Junlens, 2023-05-18:
This electric soldering iron is very useful, but there are also some inconvenient aspects to use, such as being too sensitive to induction. Even if I adjust the wake-up threshold of 50mg, the vibration of the table can put it into working mode.

Junlens, 2023-05-22:
if (abs(accel.getX() - gx) > WAKEUP_THRESHOLD && abs(accel.getY() - gy) > WAKEUP_THRESHOLD && abs(accel.getZ() - gz) > WAKEUP_THRESHOLD) {
    gx = accel.getX();
    gy = accel.getY();
    gz = accel.getZ();
    handleMoved = true;
  }
  
I made some changes and found that it works very well

Eddddddddy, 2023-05-24:
Nice, there is error with this code. But that's what open source is about.

ruevs, 2023-07-26:
It seems this https://github.com/Eddddddddy/Songguo-PTS200/commit/e40d5af15d401bb34efdaa0f05f400f47a23d09d commit fixes the problem and therefore [version 4.3.1](https://github.com/Eddddddddy/Songguo-PTS200/releases/tag/4.3.1) does as well?

Perhaps this issue can be closed?

closed 2023-09-01 by Eddddddddy