Eddddddddy/Songguo-PTS200 #9: What are these parts for?

Kabron287, 2023-07-26:
[image]

temskiy, 2023-07-27:
Слева банальная подставка под паяльник.
Справа защитный колпачок (по мнению авторов...) но он одевается только на адаптер для жал T12 той стороной где нет внутренней резьбы, а вот зачем на другой стороне резьба есть загадка. И почему бы ещё не сделать возможность одевать на колпачок для родных жал тоже не ясно.

Kabron287, 2023-07-27:
> Слева банальная подставка под паяльник.

Ну это очень сильно сказано! :))
Нужна инструкция

temskiy, 2023-07-27:
Там есть и на картинке и в тексте описано https://mysku.club/blog/aliexpress/96385.html

Kabron287, 2023-07-27:
Не описано как ей пользоваться, ибо не возможно в принципе

temskiy, 2023-07-27:
¯\_(ツ)_/¯

Kabron287, 2023-07-27:
То к темю их прижмет, то их на хвост нанижет...

Eddddddddy, 2023-07-28:
Sorry, everyone, I really don't understand Russian at all. But I may understand what you are discussing. The long part in the picture can be used as a pen cap. It can play a protective role when storing the soldering iron. The usage is to untie a part of the front end of the soldering iron and replace it with this one.

closed 2023-09-01 by Eddddddddy