Eddddddddy/Songguo-PTS200 #13: Layout offscreen english 4.5.0

resync5000, 2023-09-01:
Hi,

4.5.0 - Layout english is offscreen.

HEAT for example is cutoff.

pls fix

sansansansansansan, 2023-09-23:
Confirm, I have same issue.

hurynovich, 2023-10-07:
Same problem:

[image]
[image]
[image]

netweaver1970, 2023-11-03:
The Mandarin version of the texts are mostly very concise, 2 characters. 
And no provision was made after translation into English, to make it still 'somehow fit' with the larger English texts.
One has to modify the code to change the left position for that text a bit file and recompile+flash:
[image]

Either just reduce the 96 to something smaller, say 80 or -smarter- make the X coordinate dependent on the language chosen ...
[image]

 I might do it when I receive my soldering iron... :)

Abasz, 2023-11-21:
I confirm this is a bug in the English language for version 4.5.0.

resync5000, 2023-11-24:
thats great now that all have confirmed it, can someone please fix it?... thanks :D

dexter323i, 2023-11-25:
@resync5000  I can. :)

Pull request created.

resync5000, 2023-12-03:
> @resync5000 I can. :)
> 
> Pull request created.

thanks

Eddddddddy, 2023-12-03:
let's try 4.5.1. English has been fixed. 

because we make a U8g2 font include simple Chinese and tradition Chinese and English. Especially tradition Chinese. I use this tool. [https://github.com/createskyblue/Easy-u8g2-font-generate-tools](url). But the English letter is not monospaced font.

I restore old font for English and English letter.