Eddddddddy/Songguo-PTS200 #18: Compatibility with Xiaomi 120W charger

pricewatermouse, 2023-09-17:
Has anyone attempted to use the PTS200 with a Xiaomi 120W charger featuring a USB Type A output? 

**Does PTS200 reach the full 100W?** 

I managed to achieve a 20V output using a QC3 trigger, but I do not have the PTS200 to confirm.

[image]

pricewatermouse, 2023-10-08:
**Test results with Xiaomi 120W charger**
_Note: The charger and cable is genuine Xiaomi, from Xiaomi 11T Pro set_

- 20V(100%) - shuts down immediately when starts heating
- 20V(50%) - starts to heat for about 5 sec, then shuts down
- 15V - works ok

Enabling or disabling QC does not affect anything.
Using normal cable results in downscaling to 5V.
Using other two QC3.0/2.0 enabled chargers results in downscaling to 5V. 

Hey guys, is QC working at all?

netweaver1970, 2023-11-03:
the datasheet of the USB chip doesn't mention QC at all, only PD3.0, so I guess it will NOT support QC ... https://datasheet.lcsc.com/lcsc/2204251615_WCH-Jiangsu-Qin-Heng-CH224K_C970725.pdf

netweaver1970, 2023-11-03:
correction, the QC3 implementation is handmade, using a library and using 2 MCU pins to put/read certain data on the USB data lines to get the charging device to deliver a higher voltage.

> the datasheet of the USB chip doesn't mention QC at all, only PD3.0, so I guess it will NOT support QC ... https://datasheet.lcsc.com/lcsc/2204251615_WCH-Jiangsu-Qin-Heng-CH224K_C970725.pdf

correction, the QC3 support implementation here is handmade, using a library and using 2 MCU pins to put/read certain data on the USB data lines to trigger the charging device to deliver a higher voltage. So the USB power chip not supporting QC was a red herring.

alex-military, 2024-05-03:
Hi, I also tried this soldering iron with the latest firmware release and 120W Xiaomi charger. the result is that at 20V 100% and 20V 50% it turns off immediately when the heating starts, and restarts after about 4/5 seconds, at 15V it restarts immediately at 12V it marks the voltage as 9.8v and it works. I tried starting the heating at 12 and then going back into the menu and bringing it to maximum and it works correctly. I suspect that the heating starting curve is too steep and causes the power supply to fail. In my opinion, lowering the starting curve could work correctly. I don't know how to do it so I might ask the more experts if they can implement a 20V (smoth) function which involves a more gradual current absorption at the start and rising up to 100% once 100/200 degrees have been reached.

sbn-purchark, 2024-05-17:
> Hi, I also tried this soldering iron with the latest firmware release and 120W Xiaomi charger.

I wonder if the charger is seeing it as a short. Using a lesser/3Amp/60W cable should help reduce the surge current while still heating up plenty fast. PTS200 does not check for an e-mark chip.

Since it's simple PWM control, it may not help to slow it down. The reason it works after the tip has heated up some is because resistance increases as it gets hot.

https://www.mi.com/global/product/xiaomi-120w-charging-combo-type-a/specs/

Has an odd 11v, no 12v. At 10V, the max starting current would only be 2.5 Amp.

Instead of changing PWM, maybe it would be better to start with a lower voltage then step-up the voltage profiles incrementally. Similar to what you're doing manually.

alex-military, 2024-05-25:
sorry for the delay but I only read it now. Yes, I think that starting with a lower profile to make it increase automatically would be useful. I don't think you consider it short the only thing it does is a reboot I assume more than anything it's an overload? well, as soon as I can I'll try it with a different cable and see how it behaves. However, it would be useful to have a profile that starts with 12V and increases automatically based on a setting that can perhaps be set. because I tried with another 67W Xiaomi power supply and it has the same problem

sbn-purchark, 2024-05-25:
Forget where I read it atm, but someone mentioned that some chargers, power supplies do not like sudden/large current spikes even though they are rated to handle the max requested power. That could be what's happening here.

I got one of these ~$1 65W USB-C PD modules from [aliexpress](https://www.aliexpress.us/item/1005005810831600.html) and it worked fine with the PTS200 left at the 20V 100% setting and the USB-C cable (from Google Pixel phone) I had on hand.

A more expensive 65W module I got earlier barely worked, I had to fix the bad soldering job on it too. That $1 one is actually better quality.

Maybe those name brand adapters are being more conservative for safety/device protection reasons.

vortigont, 2024-05-26:
Hi! I've implemented PWM power ramping in my fork of this firmware, it works there. But I do not have any problematic power supplies to really test it. If anybody is willing to try it, you can find it [here](https://github.com/vortigont/ESPIron-PTS200). Any feedback would be appreciated.

sbn-purchark, 2024-05-26:
Thanks, very interesting fork, I've made a note to try it.

leocb, 2024-05-27:
Try another cable. I bought 2 new 100w Essenger cables that shows the power draw, one is 1m long the other is 2m, only the 2m works with 20v. The other 1m cable shuts down immediately when I start heating.

pricewatermouse, 2024-05-28:
LOL. It doesn't shut down because your 2m cable is crappy and adds a lot of resistance.