Eddddddddy/Songguo-PTS200 #19: The temp settings dont want to remember

sansansansansansan, 2023-09-23:
The temp settings dont want to remember. I set the 350 degrees, then Return. And nothing - the temperature settings as default...

sysytwl, 2023-09-28:
yes, i got the same problem

dexter323i, 2023-11-25:
This part causes it at line 612 in Thermostat() function:
```
if (SetTemp != DefaultTemp) {
    DefaultTemp = SetTemp;  // 把设置里面的默认温度也修改了
    update_default_temp_EEPROM();
}
```
When user finishes setting up things, and exists to main screen, then update_EEPROM() is called, which saves everything correctly. Even DefaultTemp too!
Then normal operation starts, Thermostat() called from loop, but SetTemp won't be equal to DefaultTemp, so DefaultTemp is overwitten with SetTemp, then it is also saved in eeprom. Which is wrong, because it overwrites user's previous DefaultTemp modification.

I don't know why this part is even here. What is the intended reason of it?

dexter323i, 2023-11-25:
Update: now I get it what is the reason of this part: When the user changes temp with Plus and Minus buttons, it is stored as Deafult temp, which shifts the Boost temp too.
I made a pull request to fix the original issue.