Eddddddddy/Songguo-PTS200 #21: PTS200 doesn't recognized as USB device after update

hurynovich, 2023-10-07:
After I updated firmware solder is no more recognized as USB device at all and as result I cannot flash it any more.
1. I built firmware from sources v4.5.0.
2. I installed it via menu point `Update firmware` and mounted disk.
3. Solder works but it is no more flash-able and device manger doesn't see new USB device when it connected. .

What can be the problem and how to solve it?

pricewatermouse, 2023-10-08:
You don't see "USB Serial Device (COMx)" in Windows Device Manager / Ports (COM & LPT) ?

hurynovich, 2023-10-08:
Yes, I don't see it under Windows Device Manager too.
[image]

Originally I did everything under Ubuntu/Linux.
And I also don't see USB/COM device with `lsusb` command or in `dmesg` logs.

I didn't installed any drivers. But I don't think it is required to detect USB device.

pricewatermouse, 2023-10-08:
Try another PC?

I think you require some UART to send firmware

pricewatermouse, 2023-10-08:
The ESP32-S2 chip needs to be in bootloader mode before it can be detected as a DFU device and flash. This can be achieved by pulling GPIO0 down (e.g., pressing the BOOT button), pulling RESET down for a moment, and releasing GPIO0.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/dfu.html

**GPIO0 is connected to "Confirm" button of the PTS200** 

**Flash Download Tools**
https://www.espressif.com/en/support/download/other-tools

hurynovich, 2023-10-08:
Yeah, I also thought about this button.
But still trying to disassemble the solder somehow.

pricewatermouse, 2023-10-08:
Just hold "Confirm" (middle) button and insert usb, then ESP32-S2 (USB\VID_303A&PID_0002&REV_0723&MI_02) device appears. After installing zadig

pricewatermouse, 2023-10-08:
Simply press and hold the "Confirm" button (located in the middle), then insert the USB cable. This action will cause the ESP32-S2 (USB\VID_303A&PID_0002&REV_0723&MI_02) device to appear. 

Once you have installed the [Zadig tool](https://zadig.akeo.ie/) (on a Windows system), you will be able to use a USB serial device to upload firmware.

hurynovich, 2023-10-08:
Ah, got it now.
I managed to upload firmware.
@pricewatermouse Thanks a lot!

hurynovich, 2023-10-08:
> Simply press and hold the "Confirm" button (located in the middle), then insert the USB cable. This action will cause the ESP32-S2 (USB\VID_303A&PID_0002&REV_0723&MI_02) device to appear.
>
> Once you [...]

Ah, got it now.
I managed to upload firmware.
@pricewatermouse Thanks a lot!

closed 2023-10-08 by hurynovich