Eddddddddy/Songguo-PTS200 #22: How to disassemble PTS200?

hurynovich, 2023-10-07:
I am not sure if it a right place to ask.
But how to disassemble and extract circuit board from solder washout damaging it?
May be do you have photos of how board is mounted inside the body? 
I have such one:
[image]

pricewatermouse, 2023-10-08:
Put with buttons down, push with usb connector then pull the board with tweezer?

hurynovich, 2023-10-08:
I didn't managed to pull it out. It feels like monolith.

Does it have some lock mechanism? 
By pressing buttons down and pushing with USB connect I didn't managed to move it at all.
And it feels like buttons doesn't block circuit board.

It also looks like cap from behind is removable.
How is it mounted? Threading, pressing, glue?
[image]

pricewatermouse, 2023-10-08:
Here is a video from Eddddddddy_Wang that can provide you with some insights into board design. However, it doesn't appear to be the final version

https://www.bilibili.com/video/BV1os4y1u7qj/

mptei, 2023-11-04:
Opening is very easy. You need to remove the display foil. To do that warm it up a little bit and then use a plectrum to take it off.
Below you find the screw which holds everything together. You need a tiny screwdriver.

hurynovich, 2023-11-04:
Thanks, for info.
Seems like it requires 0.9 mm hex screwdriver.

closed 2023-11-04 by hurynovich