Eddddddddy/Songguo-PTS200 #23: QC 不能触发

sysytwl, 2023-10-16:
我试了好几个充电器，换了好几个线。都不行。
hit：QC3的库是基于5v的arduino做的

netweaver1970, 2023-11-03:
as per the readme here from Eddy Wang, the USB chip is the CH224K. and in the official English datasheet I find this: 
[image]

Maybe the PTS200 V2 has a different/better USB chip but there is not information on it.

Jiawens, 2023-11-03:
According to `SolderingPen_ESP32S2.ino`:14 :

```
QC3Control QC(14, 13);
```

It seems that QC is implemented by using ESP32S2's GPIO w/o CH224K.

netweaver1970, 2023-11-03:
> According to `SolderingPen_ESP32S2.ino`:14 :
> 
> ```
> QC3Control QC(14, 13);
> ```
> 
> It seems that QC is implemented by using ESP32S2's GPIO w/o CH224K.

yep, I saw that later too. I was too trigger happy. 
it seems only used for USB-A connectors, as per the library readme: 
[image]

https://github.com/vdeconinck/QC3Control

So I have no clue yet why it doesn't work, when I get my own soldering iron, I'll start digging around to see what does/doesn't work. A schematic might help too, or an easy way to disassemble the iron.

netweaver1970, 2023-11-03:
indeed, the D+ and D- are connected through otherwise firmware upload would not work. 
I guess the statement in that readme was more a normal physical 'source' plug statement than a wire protocol statement with a bit limited wording.
If you start with an usb-c at the charger side, than PD is a more likely solution than QC. But technically both could work side-by-side, if the source supports also QC over usb-c. It's not the most common solution though. Of course only 1 solution needs to be selected by the target device (the iron).

In 2 weeks time I'll also have that iron. It's pity, I thought I had bought just a new iron with potential, for a good price, only to be used in Singapore when visiting the family. As a part of my remote mini workshop, so to speak.
Now it seems I have even bought a new project by itself, as if I don't have enough active projects yet :D

https://github.com/Eddddddddy/Songguo-PTS200/issues/22 also has issues fully disassembling it.

sysytwl, 2023-11-04:
https://github.com/vdeconinck/QC3Control#how-to-connect

QC3.0 lib requires an external circuit setup. and it is only for 5v Arduino (check the resistance).

> When the Arduino is in the booting stage, all the pins are floating. So, the output voltage of the circuit is 0.6v,
> Then pull down. "Acknowledge"
https://github.com/vdeconinck/QC3Control/blob/51d640959f92ff5d2d63cb4d02b39fc6a09d4a4e/src/QC3Control.cpp#L28
> If I want 12v, D+ and D- should be 0.6v again. 

- my charger 
[image]
- solder pen **12v !!!!!!!**
[image]

[image]

- a 9v power bank
[image]

[image]

### This is the worst product I bought in my life. Only hardware, no software.  I think Arduino IDE for ESP is only for universities.
### meanwhile, I highly doubt the cable they sold. If this is a USB 3.2 Gen2 cable, where are the pins for USB3? Or they just randomly wright something in the e-marker. So, can I trust the 50V and 5A below? 

[image]

sysytwl, 2023-11-04:
> indeed, the D+ and D- are connected through otherwise firmware upload would not work. I guess the statement in that readme was more a normal physical 'source' plug statement than a wire protocol [...]

yep, this function is not possible. QC3 or data transfer. Unless the pin is a DAC pin.  I'll use another person's firmware now.

Eddddddddy, 2023-12-03:
> 我试了好几个充电器，换了好几个线。都不行。 hit：QC3的库是基于5v的arduino做的

> Hi, Interesting question :-) **Disclaimer:** I have not tested anything of the following. I would try with those values: R1 = R3 = 4700 Orm R2 = R4 = 1500 Ohm R5 = R6 = 100 Ohm This is just the result of some fiddling with the values in the blue cells of the spreadsheet available in the blog post and adapted for variable values of Vcc, attached here: [Resistor calculator variable Vcc.zip](https://github.com/vdeconinck/QC3Control/files/5490865/Resistor.calculator.variable.Vcc.zip) The goal is to bring the 5 big yellow dots close to green dotted lines, each in its own area (delimited by black lines) :-) I guess you can try those values and see how it goes. It shouldn't break anything (although I take no responsibility as once again this was not tested). Don't hesitate to report your results :-)
 [https://github.com/vdeconinck/QC3Control/issues/13#issuecomment-722019268](url)

QC3 still in beta. In some cases it will conflict with the ESP32's USB communication. Therefore, in the software settings, this feature is turned off by default.

For 3.3v vbus information, you can view the link above.