Eddddddddy/Songguo-PTS200 #35: blank display

Cory844, 2024-01-01:


Cory844, 2024-01-01:
/trying to update firmware think i erased everything , any fix for this

sbn-purchark, 2024-01-01:
See https://github.com/Eddddddddy/Songguo-PTS200/issues/34. I had the same problem, I managed to get 4.5.1 user.bin on which got it working again.

Cory844, 2024-01-01:
how did you flash it ? i cant even get the computer to recognize it unless i hold int he middle button while powering it on

sbn-purchark, 2024-01-02:
You may still need to hold down the middle button when powering it on since the firmware on it is broken. I forget if I had to do that or not. Maybe that button is tied to the boot mode of the ESP chip, putting it into programming/flash mode.

sbn-purchark, 2024-01-02:
If it still has issues, try putting on 4.5.0 (factory or user whichever works), then put on 4.5.1 user.bin. Since my PC was acting up, I'm not exactly sure what finally did the trick.