Eddddddddy/Songguo-PTS200 #36: Power supply

Cory844, 2024-01-02:
What power supply does ever one use ? Will a laptop charger work ?

sbn-purchark, 2024-01-02:
I'm currently using a 45W chromebook (USB-C PD) adapter. I have to limit the voltage to 15V (in the settings) to prevent it from cutting out (the PD adapter I have can do 20V, but only @ 2.25Amp). If it is a USB-C adapter meant for a laptop then it will probably have the required power.

It needs 5A @ 20V to get the max 100W capability.

It's also possible to power from a bench supply with the right adapter. I haven't come-up with verified details on this yet, such as the maximum voltage rating.

sbn-purchark, 2024-02-06:
I have yet to come across the max working voltage. If it is set to 20V 100%, it looks like it won't limit the power. So you must keep the voltage at 20V at the unit when heating or risk burning up the heating element.