Eddddddddy/Songguo-PTS200 #39: Compiling 4.5.3

laverdanny, 2024-02-10:
hi
your projet is great !
I wanted to add two languages (French & Italian), but compiling fails

I have several questions about compiling under Adruuino 1.8 or 2.3
1) Is the file /boards/PTS200.json to be included in Arduino IDE ? If YES, how to do it ?
2) PID_v1 is not in Aduino library list, I took in in https://github.com/wagiminator/ATmega-Soldering-Station/tree/master)/Arduino-PID-Library.zip. Is it right ?
3) I tried both IDEs, compiling stuck on 
C:\Users\Dan\Documents\Arduino\SolderingPen_ESP32S2\SolderingPen_ESP32S2.ino: In function 'void setup()':
C:\Users\Dan\Documents\Arduino\SolderingPen_ESP32S2\SolderingPen_ESP32S2.ino:168:10: error: 'class USBCDC' has no member named 'setRxTimeoutMs'; did you mean 'setTxTimeoutMs'?
   Serial.setRxTimeoutMs(0);
          ^~~~~~~~~~~~~~
          setTxTimeoutMs
4)I read somewhere it is possible to upload USER.BIN or FACTORY.BI§N, but I haven't found them

laverdanny, 2024-02-10:
About 3), I changed to setRx just to try to compile, reverted back to setTx...  and now it compile ! Don't mind about 3) anymore

5) are you interested in getting French & Italian ? how to send to you ? Regards

sbn-purchark, 2024-02-11:
This project is setup to use Platform IO, that would make things easier. It also specifies the library versions being used.

https://github.com/Eddddddddy/Songguo-PTS200/blob/main/platformio.ini

boards file is also for that.

BINs are in releases, https://github.com/Eddddddddy/Songguo-PTS200/releases

https://github.com/Eddddddddy/Songguo-PTS200/blob/main/docs/update_guide.md mentions the difference between user/factory.

Regarding PIDv1, this is mentioned in the code:
```
#include <PID_v1.h>  // https://github.com/wagiminator/ATmega-Soldering-Station/blob/master/software/libraries/Arduino-PID-Library.zip
// (old cpp version of
// https://github.com/mblythe86/C-PID-Library/tree/master/PID_v1)
```

The links for the other libraries are in there too.