Eddddddddy/Songguo-PTS200 #41: Turns itself off and restarts after reaching set temp

myconite, 2024-05-19:
The video shows the solder pts200 turns itself off and restarts after reaching set temp. It should stay at that temp, not restart. It is not normal behaviour. I don't suppose there's a hidden setting I forgot to set? Otherwise it seems some of the components might be faulty.

[https://youtu.be/f8GsFNOK_b0?spm=a311a.7996332.0.0.6a147421YC139s](url)

sbn-purchark, 2024-05-19:
I noticed it shuts down at 375 with it set to 380.

- What happens if you try to immediately reheat?
- What happens if you lower the temperature setpoint?
- Try lowering the voltage in settings.

What's your power source?

myconite, 2024-05-19:
Ah! Thanks a lot for the second pair of eyes, the situation is solved. I just received this solder and tried it for the first time. I will answer all your questions if anyone has the same issue in the future.

1. If I try to immediately reheat, the situation repeats again. Solder restarts.
2. Lowering or increasing the temperature did not make any difference in my former tests.
3. Voltage was the problem. Even though my 65W charger could support 20V, it appears this was the issue. I had to lower the voltage to 15V and the situation was resolved. 

Thanks anonymous friend :)