Eddddddddy/Songguo-PTS200 #44: Please add detailed build decumentation

n3roGit, 2025-01-21:
Unfortunately I can't get the part built. Can you perhaps write detailed instructions in the README?

`/GitHub/Songguo-PTS200/SolderingPen_ESP32S2/SolderingPen_ESP32S2.ino: In function 'void PD_Update()':
/GitHub/Songguo-PTS200/SolderingPen_ESP32S2/SolderingPen_ESP32S2.ino:1493:5: error: 'ledcSetup' was not declared in this scope
 1493 |     ledcSetup(CONTROL_CHANNEL, CONTROL_FREQ_20V, CONTROL_RES);
      |     ^~~~~~~~~
/GitHub/Songguo-PTS200/SolderingPen_ESP32S2/SolderingPen_ESP32S2.ino:1495:5: error: 'ledcSetup' was not declared in this scope
 1495 |     ledcSetup(CONTROL_CHANNEL, CONTROL_FREQ, CONTROL_RES);
      |     ^~~~~~~~~
/GitHub/Songguo-PTS200/SolderingPen_ESP32S2/SolderingPen_ESP32S2.ino:1498:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
 1498 |   ledcAttachPin(CONTROL_PIN, CONTROL_CHANNEL);
      |   ^~~~~~~~~~~~~
      |   ledcAttach
*** [.pio\build\solderingpen\src\SolderingPen_ESP32S2.ino.cpp.o] Error 1`

ewpa, 2025-01-21:
Looks like you'll need to compile using ESP32 Arduino release earlier than v3, e.g. v2.0.17.

n3roGit, 2025-01-22:
[image]

I tried the Arduino IDE and VSCode with PlattformIO. Both the same.

Abasz, 2025-01-22:
Can you please post the part of the output of pio run which has the title `PACKAGES:`

I suspect that pio has previously installed a newer arduino and arduino-espressif version and now it tries to use that. You need to clear .platformio folder before building. Reason is that if a newer version of arduino framwork (3.x) was installed in the past this pio will use that instead of the needed 2.x version.

Please remove `.cache`, `packages`, `platforms` folder and rerun `pio run`

This is what should be shown under PACKAGES
```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
```

ewpa, 2025-01-22:
If using Arduino, under boards manager, ESP32 should be something like this:

[image]